rhysmorgan134/JaguarXf_CAN #2: OBD2 pinouts in the readme incorrect

alierhan, 2020-04-09:
In the readme pinouts specified as pin 3=CAN-H and pin 11=CAN-L

Standard pinout is pin6=CAN-H and pin 16=CAN-L
[image]

rhysmorgan134, 2020-04-09:
For the program to work, you need to be onto the medium speed can bus. This is found on pins 3 and 11, however, this is jaguar specific and may change for other manufacturers. I will put the fact its for medium speed in the readme to make this clearer.

closed 2020-04-09 by rhysmorgan134

alierhan, 2020-04-09:
> For the program to work, you need to be onto the medium speed can bus. This is found on pins 3 and 11, however, this is jaguar specific and may change for other manufacturers. I will put the fact its for medium speed in the readme to make this clearer.

Cheers mate