rhysmorgan134/JaguarXf_CAN #3: CAN-L unable to communicate

alierhan, 2020-04-09:
Hi Rhys,

I can see data coming from CAN-H when the engine is running, but unable to get any data with bit rate 125000

My /boot/config.txt
#CAN bus controllers
dtoverlay=mcp2515-can0,oscillator=16000000,interrupt=25
dtoverlay=spi-bcm2835-overlay
How to specify can1 to be used for CAN-L? Is there info I need here to get hardware to expose can1?

sudo /sbin/ip link set can0 up type can bitrate 500000
brings up the port and works fine

sudo /sbin/ip link set can1 up type can bitrate 125000
returns Cannot find device "can1", I'm assuming there's a configuration entry I am missing.
My plan is to have both CAN-H and CAN-L running so that I can get data from both ports of the PiCAN2. I hope this is possible

Cheers

rhysmorgan134, 2020-04-09:
Hi, 

For sure its wired onto the wrong pins on the OBD connector, the medium speed and high speed are two seperate circuits entirely, pins 11 and 3 are the ones required for medium speed connection, see the attached picture for clarity :)

[299361954-Electrical-Wiring-Diagram-for-jaguar-xf-250.pdf](https://github.com/rhysmorgan134/JaguarXf_CAN/files/4453747/299361954-Electrical-Wiring-Diagram-for-jaguar-xf-250.pdf)

rhysmorgan134, 2020-04-09:
After reading one of your other comments, I am not sure if you are using this on a jaguar XF or if you are looking to customize it for another car. If its not a Jaguar XF then double check which pins are for medium speed, as they certainly do vary between manufacturers.

alierhan, 2020-04-09:
> After reading one of your other comments, I am not sure if you are using this on a jaguar XF or if you are looking to customize it for another car. If its not a Jaguar XF then double check which pins are for medium speed, as they certainly do vary between manufacturers.

I am using it on Jaguar XF 2009, so I rewired my OBD2 will test again soon

closed 2020-04-09 by rhysmorgan134