rhysmorgan134/JaguarXf_CAN #4: Installation instructions

alierhan, 2020-04-09:
I would like to use your JaguarXF_CAN as a base for my project, could you please provide some instructions on how to get this installed and run as an app on Open Auto.

Not very proficient with this type of software development, your help will be very valuable.

Thanks

rhysmorgan134, 2020-04-09:
Hi,

Firstly to run the software, its as simple as double clicking the Jag-HU.AppImage file in "install" directory, aslong as you have a can0 connection it should load straight up.

For the open auto side, all thats needed is to add this .AppImage file to the external apps list.

If you still need help, let me know and I will do a write up on it (a more in depth guide is coming on software side)

alierhan, 2020-04-09:
> Hi,
>
> Firstly to run the software, its as simple as double clicking the Jag-HU.AppImage file in "install" directory, aslong as you have a can0 connection it should load straight up.
>
> For the open [...]

Most of the things are working now, at least I now have a good example to work with. 
I wasn't able to get any data from the candump, a bit more investigation from me is needed I think, I'll check my setup read your code and see if I can find the reason.

Thank you so much mate

rhysmorgan134, 2020-04-09:
Good stuff, and thanks for giving it a go! Your input will be mega useful!

closed 2020-04-09 by rhysmorgan134