rhysmorgan134/JaguarXf_CAN #5: Screen dimming

alierhan, 2020-04-09:
My XF had headlights on AUTO, when I started the car headlights were on, I've noticed that pi screen dimmed a little. I could control amount of dimming with the usual pot under the steering wheel. When I moved the light switch to OFF, pi screen went completely blank, manual adjustment did not work at all.

I do not have daytime running lights, not sure if that could be the reason.

alierhan, 2020-04-10:
I will check out the dimming level when I take her out for a drive.

I started going through the code, made some changes to index.html as trying to test and align the objects etc. I then realised that there was an error

Loading failed for the <script> with source “file:///socket.io/socket.io.js”.
app.js was unable to locate socket.io.js then I found that the repository does not have node-modules folder. Now, being unfamiliar with java/html, remembering how nam was used in the tutorial to add socketcan package. I checked the package.jason, there was some entries but nothing for socketcan. I decided to install it with nam, this probably will make it work. Let me know if I'm on the right track.

70 years old means it's very difficult to remember the stuff, especially these apps have objects and code in separate files and folders, very challenging for me. I guess good for dementia :-)

rhysmorgan134, 2020-04-11:
sorry for the late reply! if you are making changes, firstly make sure you are on the v1.01 branch so:

```
git clone <url>
cd JaguarXf_CAN
git checkout v1.01
cd electron/app
npm install
sudo ./node_modules/.bin/electron-rebuild
```

with all of that done, you should be able to type

`npm start`

It will then open a development version of the app using the files directly hosted by the server. The reason you have so much missing (ie socket.io) is because the server normally servers these up. But if you just open the index.html, then you dont have any of the other resources running.

let me know how it goes!

And hats off to you doing this at 70! I hope im as active into this stuff when im that age!

rhysmorgan134, 2020-04-11:
just to add the reason you were missing socketcan (and all other modules) is that when you commit to git, you dont include the node_modules folder (for a variety of reasons, sheer amount of files is one)

So when you clone the project, you have to run npm install in the directory, this then looks at everything in the package.json, downloads, and then installs all modules.

alierhan, 2020-04-14:
After cloning version v1.01 I believe the dimming is still an issue for me.
When I run the app on the bench no OBD2 available, I had blank screen
I decided to run it connected to XF, again blank screen
I started the engine no change, dimmer switch makes no difference
Headlights were on at the time

I'm going to be off-line for a while to rebuild my Raspberry PI, possibly my MacPro as well due to corruption issues. I may need to disable dimming when I'm ready to start again.

New layout looks great, one suggestion, for fuel consumption, I'd like to see instant consumption as well as average, best way is to display instant fuel consumption when the car is moving, switch to average when it's stationary, let me know what you think.

Cheers

alierhan, 2020-04-14:
The data coming from steering wheel buttons to control the audio, should be the same as data from the audio control panel. Am I right?

alierhan, 2020-04-14:
I probably will use this image instead of layout of the car. Not sure about copyright issues etc though if it's made publicly available.
[image]

rhysmorgan134, 2020-04-21:
Sorry for the late response, so to cover the points:

- For dimming, I suspect you may have a faulty solar sensor, or perhaps this is an optional extra on the cars and maybe yours is fitted without? When you are back online, I will send you the can id and byte for the solar sensor, you can give it a test with can sniffer then, just covering it up with your hand makes the values change. Either way I will add an option to turn auto dimming off.

- For the image, the XF is there, as the plan is to dim out the background and overlay the parking sensors on there when they are active. But for now feel free to change to what you need, just replace the image file, and then rebuild the app.

- The steering wheel control are done using an arduino micro, unfortunately they arent on CAN, but give an analog value direct to the IAM, so its a case of tapping into that, then the arduino converts the button presses to keyboard presses.

rhysmorgan134, 2020-04-21:
would be interested to see what you mean by blank screen too, is it blank as in too dark? or blank as in nothing loads?

alierhan, 2020-04-21:
Hi mate,
I spent last few days capturing candump data trying to understand better. I imported the logs into spreadsheets, which allows me to have summary data as well as the whole capture. 

If I run V1.1 the backlight is at lowest level, with the previous version backlight is on. This is what I meant by blank screen. From my previous obd2 scans I remember seeing a fault related to the light sensor. Not worried about it as AUTO setting on the headlights has been working fine. I pay more attention to the sensor. If you could let me know the id of the sensor, I should be able to see what’s going on easily.

Picture is just a novelty.

Steering wheel control is important for me, with no iPod or USB connection available, controlling media from those controls is desirable. I’ve done few projects with arduino mini, one being reversing start/stop behaviour on our Ford Focus. I’d be interested tackling this if more information is available.

Best regards,
Ali Erhan
0414 760780

alierhan, 2020-04-21:
Do you have a document available explaining module id’s and data Goes the canbus? I think it will be useful going deeper into this project.

A Jaguar training material came to my attention for older jaguars. I had a quick browse through Jaguar training material regarding canbus modules and data being sent through the bus. There’s a notification saying that modules can be different in various markets around the world, local dealers will have specific information.

Given that older jaguars didn’t  had to confirm global standards, I’d take this with a pinch of salt.

Best regards,
Ali Erhan
0414 760780

rhysmorgan134, 2020-04-22:
Ok so the can I'd is 0#168 and it's byte 3 (remembering that the bytes start at 0) that should be  the data for solar sensor (one mounted in centre of dash) 

You may know already but if you start cansniffer and type:

-0000

That removes all IDs then type +168 will add only that ID in.

Il get some information written up on the arduino

Let me know what you get back from that, like I say covering the sensor should immediately trigger a change in values (lights may need to be off)

alierhan, 2020-04-22:
From my can dump data from previous test for module id 168 I have the following:
3 x 0000000000000000
16 x 000000D400000000
930 x 000000D500000000
512 x 000000D600000000

The duration of the test was approx 2 minutes

First 000000000 appears as first thing possibly repeats every minute, therefore should be ignored to my opinion.

D4, D5, D6 should be normal as it was getting darker. I will be doing another test soon, I’ll update my results and get back to you.

Please see my document attached for known modules. let me know if it’s good enough.

Ali Erhan
Photographer

0414760780
ali.erhan@me.com

alierhan, 2020-04-22:
Hi Rhys,

Here’s the data I’ve got for can bus id 168

without light sensor covered 3rd byte varied 80-85
with light sensor covered i’ve got 87-e9

headlight switch was OFF

Cheers,

Ali Erhan
Photographer

0414760780
ali.erhan@me.com

alierhan, 2020-04-24:
Here's the spreadsheet document for the modules I know about.

[000 KnownModules.ods.zip](https://github.com/rhysmorgan134/JaguarXf_CAN/files/4526011/000.KnownModules.ods.zip)

alierhan, 2020-04-29:
Hi mate,
Today I captured all the data from ID=2C8 as it seems most information is contained here. I updated the Decoded.json with my test results, some information conflicted with the old json this one is correct for what I've seen from my tests.

Hope you're safe and well.

Cheers
[MS_CAN_Decoded.json.zip](https://github.com/rhysmorgan134/JaguarXf_CAN/files/4550285/MS_CAN_Decoded.json.zip)

rhysmorgan134, 2020-04-29:
Looks pretty accurate, I'm off for a few days now so a few things I need to do, clean the repo is one. That file that you have used is old, theres actually a direct json that the app loads in on start up that contains all the IDs

https://github.com/rhysmorgan134/JaguarXf_CAN/blob/master/electron/app/resources/canOut.json

the canMap one is for the feed back from the lights.

I will also look at implementing local storage so an option to disable automatic dimming can be added. Finally I will put in a diagnostics page, that gives read outs of values, so we can perhaps look at fault finding the auto brightness on your set up (will have a raw read out of the sensor value, then also the scaled value of the read out)