rhysmorgan134/JaguarXf_CAN #7: Detecting power off via can bus

doctordbx, 2020-05-21:
Great work so far Rhys.

I Just installed the screen today and am waiting on my can module to do the can work. I am wondering if there is a method for detecting ignition off via the can bus to trigger a shutdown of the pi device? Is this something you have looked at?

[image]

alierhan, 2020-05-22:
Hi mate,
Looks very neat indeed.

I found an used plug under the double cup holder, which has 3 terminals. One is ground, one is 12v and the third one is used for dimming cabin lights. If you get 12v from this plug, it can be used to trigger a clean shutdown of the Pi. 12v from this plug goes off when the doors are locked.

There are readily made devices to shutdown Pi when power off is detected. I am powering the Pi from Canbus 12v which is always on, I'm going to use the second 12v power from the floor console to trigger the shutdown.

Cheers,
Ali

doctordbx, 2020-05-22:
Ah cool. Would be nice to get an idea of how you've wired it up. Are you
taking CAN from the OBD plug or from the control socket for the panel we
removed?

I haven't got my pican yet but wired up an esp32 with a can transceiver
pumping can signals out over wifi but the wireless is very patchy in the
car. So might defer any further work there until I get the hat.

Also an idea how your wired up the serial for the steering controls, but
not a big one for me as I find it's ok with Android audio and Bluetooth to
the car. Really open auto Pro is just a Android auto host for me.

For the mounting I made a 3d printed chassis with ears similar to the
mounts for the panel we removed.

alierhan, 2020-05-23:
For the details of how the wiring is done, check out these projects: https://www.instructables.com/howto/pi+shutdown/

Just replace the physical button with a 12v relay or a small arduino mini or something and that should do the job, my preference is https://www.instructables.com/id/Simple-Raspberry-Pi-Shutdown-Button/

For steering controls, I am waiting for Rhys to send me some information. This is important for me.

I would like to understand how you made the 3d printed mounting chassis. I don't have a 3d printer but I should be able to get one done for me.

All the best, let us know how you go with the shutdown and also the pi hat.

Cheers

rhysmorgan134, 2020-06-05:
Sorry guys, I'm so late to the party, things have been manic here

@alierhan I will send add the arduino code and some details today, I promise.

@doctordbx that looks awesome, would love to have an stl of the 3d printed ears!

rhysmorgan134, 2020-06-05:
For the power off issue, its only my list, but not high.

Currently I have the power button on the button strip under the screen wired to the power down of the pi, I just make a habit of doing that for now until I can get more into it.

If you want to take a stab, take a look at this

https://bluewavestudio.io/community/showthread.php?tid=1128

another option we can do is to just create a condition where if no can signals go the the pi for x seconds it shutsdown, and have it connected to permanent 12v, not sure what standby current is on the pi though...

doctordbx, 2020-06-06:
> not sure what standby current is on the pi though...

Apparently pretty terrible. Found a device called Sleepy Pi... but a bit on the $$ side for what it is.

https://spellfoundry.com/product/sleepy-pi-2/

Good to see you're still alive and kicking BTW Rhys. was worried for a bit there.

doctordbx, 2020-06-06:
Could probably make something with a relay hooked up to a micro board that can also direct the pi to shutdown and then power off after 30 seconds.

rhysmorgan134, 2020-06-06:
Yeah there's a few options, have also looked at the sleepy pi.

I'm considering using an Arduino with a can hat to control power up/down via the can. I do believed these can be super low voltage.

doctordbx, 2020-06-06:
Actually, since I ended up building my own CAN Adapter with an ESP32 and send the can signals out over WiFi through a UDP socket it occurred to me that I could just use this microcontroller and a suitable relay to do the job. I can even tell the PI to shutdown over the UDP socket and then flip the relay 30 seconds later. Might take a few days before I circle around back to that one.

alierhan, 2020-06-06:
Hi Rhys,

Sorry mate,
Just finishing off another project a dashboard for my home automation.

I’ve been thinking of using a small arduino board as you mentioned powering it up from odbc, which is 12v always available, I can tackle this easily, just give me few weeks. I’ll start on this as my next project.

I’ll let you know how I get along.

Best regards,
Ali Erhan
0414 760780

rhysmorgan134, 2020-06-06:
Awesome, thanks @alierhan . SWC files are in, incase you missed the message. I've done a quick write up in this issue #5 

this is what its based on
https://www.youtube.com/watch?v=Caf-yBI9rHA&t=309s

alierhan, 2020-06-12:
Hi Guys,

A quick update on safe shutdown for Raspberry Pi, I have a working model on the bench with few small issues to sort out.

In the heart of the setup is a deli timer I bought on eBay
https://www.ebay.com.au/itm/DC-12V-LED-Display-Trigger-Cycle-Delay-Timer-Switch-Turn-On-Off-Relay-Module/184078100906?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649 <https://www.ebay.com.au/itm/DC-12V-LED-Display-Trigger-Cycle-Delay-Timer-Switch-Turn-On-Off-Relay-Module/184078100906?ssPageName=STRK:MEBIDX:IT&_trksid=p2057872.m2749.l2649>

I have the timer is adjustable, its set to 25 seconds for now.

There are 2x12v feeds, one from can bus (always on) second one is from ACC (cigar lighter socket, turns off when the doors are locked).

When the doors unlocked Pi will get power and start booting. at this time Pi gets power from the cigar lighter.

When the doors locked, power to cigar lighter goes off, timer starts to count down, sending 12v from the canbus during this time.

When timer reaches 0, power is completely removed from Pi (12v to 5v converter).

The timer stays on all the time receiving 12v from canbus, I measured the current it is drawing 50ma on standby I think it is acceptable.

The problem:
There is a voltage drop off during the switching from one source to other. I need to do more tests on that and discuss with few people as my skills on this area is limited.

I keep you posted how I get on.

Next step is to document the full thing and share.

Cheers,

Ali Erhan
Photographer

0414760780
ali.erhan@me.com

alierhan, 2020-06-15:
Hi,
After going through few options, I decided simple is the best.

**Python Code**
In the attached zip file you will find python code for Raspberry pi, allows pi to safely shutdown and reboot.
I used 2 pins for reboot and shutdown
the other 3 pins are optional used to show the status of buttons and Pi itself

**Arduino sketch**
The sketch for arduino to control the power feed to the Raspberry Pi. I used UNO board for testing as I don't have NANO at the moment. I will possibly use UNO as I can use the same board to capture steering buttons and communicate with the Pi. This sketch monitors ignition power through a 12v relay, which activates UNO pin2. When pin2 is deactivated (ignition OFF) UNO goes to sleep mode to save power. It wakes up when ignition is ON.
2x5v relays are used to provide +12v power to the 12-5v DCtoDC converter and simulate 3 second button press for the shutdown button for Pi

I made a small board with a reboot button and an RGB LED to show button status as well as Pi status.

Now I will go to the next stage to run the entire system with PiCan hat and document the details not covered here. Please let me know if anything not covered. I'll get back with the rest within a week.

Cheers

[safeShutdown.zip](https://github.com/rhysmorgan134/JaguarXf_CAN/files/4779500/safeShutdown.zip)

rhysmorgan134, 2020-06-15:
Thanks @alierhan, I will take a look.

Something worth bearing in mind. I've started looking at making a canblocker/repeater to allow the final replacement of the OEM screen. I think I am going to use an ESP32, with a dual canbus card (I have one and have been testing some basic stuff) it may be worth thinking of integrating into this. 

My initial thoughts are along the lines of, sending the ESP32 in sleep mode when no messages arrive for x amount of time. And then setting the interrupt up, to wake up when a can message is received, which in turn powers on the pi.

It would be a nice two birds one stone.

As an aside, which cigar lighter supplies power when the doors are opened? My arm rest one is only when ignition is active, (I dont have the one in the 3rd cup holder, so perhaps its this one. or maybe its a regional difference)

rhysmorgan134, 2020-06-15:
this is the card I am using hooked upto the ESP32

https://www.waveshare.com/2-ch-can-hat.htm

alierhan, 2020-06-15:
I’ll order a esp32 next, it should work as it seems the library I’m using is tested on esp32. However, timed and 

My cigar lighter works with the doors. There’s also a 3way socket under the double cup holder seems to be on the same circuit. Possibly it’s there for the other outlet used to be in the single cup holder. Ignition can work as well it just few extra seconds until Pi boots up. Alternatively you can hook it up to cabin light circuit somewhere before the dimmer.

That canpi hat is reasonably priced, I paid premium for mine. Is 2 channels required for high speed canbus as well as medium speed?
From the specs of this board, I will have to relocate the shutdown pin on Pi to avoid interference. I’ll update the code.

What interface are you using for esp32/Pi wifi or Wired Software serial Tx/Rx?

Best regards,
Ali Erhan
0414 760780

rhysmorgan134, 2020-06-15:
> My cigar lighter works with the doors. There’s also a 3way socket under the double cup holder seems to be on the same circuit. Possibly it’s there for the other outlet used to be in the single cup [...]

Im thinking of leaving the pi permanently connected. The pi4 has greatly improved its standby current, apparently as low as 3ma. Then just using the esp to pull down global en to boot up.

> That canpi hat is reasonably priced, I paid premium for mine. Is 2 channels required for high speed canbus as well as medium speed?
> From the specs of this board, I will have to relocate the shutdown pin on Pi to avoid interference. I’ll update the code.

The two channels will be so that the IAM module will sit on a single channel with just IAM and the ESP, the other channel will be on the medium speed, and relaying all messages to the IAM. Messages out of the IAM will get filtered then relayed to the medium speed network.

> What interface are you using for esp32/Pi wifi or Wired Software serial Tx/Rx?

To be determined, I may even just use CAN as its already present.....

doctordbx, 2020-06-16:
I am using the ESP32 with a CAN Transceiver and sending the CAN bus data
out over Wifi.

I use Canneloni as the transport for Wifi.

https://awesomeopensource.com/project/mguentner/cannelloni

Using the ESP32 as a CAN Proxy is a good idea but I wonder about the
processing power.

I'm not sure the top screen is really possible to replace without getting
access to the MOST bus.

rhysmorgan134, 2020-06-16:
Canneloni looks cool. Will definately take a look. 

> Using the ESP32 as a CAN Proxy is a good idea but I wonder about the
processing power.

Yeah I have some concerns on this too, out of the main options I saw, the ESP caught my eye due to the dual core, being able to assign certain functions to certain tasks. Worst case if it can't keep up, I will just have to figure which IDs the IAM is actually looking for, and set the hardware filters for that.

> I'm not sure the top screen is really possible to replace without getting
access to the MOST bus.

I believe it can be done, the screen may need relocating to the glove box, as various other cars require (soley for the source select, however this may still work via canbus. Theres most likely other issues, but currently this is all I can forsee)

doctordbx, 2020-06-16:
Hrmm I think instead of doing the top screen I might tackle the dash next

https://www.aliexpress.com/item/4000575737861.html?spm=a2g0o.detail.0.0.b83e76acnZv4en&gps-id=pcDetailFavMayLike&scm=1007.12873.140318.0&scm_id=1007.12873.140318.0&scm-url=1007.12873.140318.0&pvid=fab37958-191e-48a7-af4b-5407b6e3ac91&_t=gps-id:pcDetailFavMayLike,scm-url:1007.12873.140318.0,pvid:fab37958-191e-48a7-af4b-5407b6e3ac91,tpp_buckets:668%230%23131923%2313_668%23808%233772%23916_668%23888%233325%238_668%232846%238107%2354_668%232717%237565%23768_668%233164%239976%23331