rhysmorgan134/JaguarXf_CAN #8: Car not responding to Can signals plus top panel buttons

doctordbx, 2020-05-30:
So I finally got my can bus hooked up and reading CAN signals in fine. However, when I send can signals for the climate control down to the car it doesn't respond.

My thoughts around this are possibly because the car is 2013 plate XF they might have changed the CAN BUS ID of the climate control module and hence it is not responding to the packets.

Rhys do you have the pinout for the climate control module to save me time having to put it back in the car and try and log the signals again? That way I can just set it up on the bench and get them exactly.

What voltage is the unit and what are the CAN pins?

Also I note some of the top buttons don't work and I think maybe because they are linked to the climate control panel. Is this the case on yours as well?

doctordbx, 2020-05-30:
Rhys I just looked at your video where you show your panel and now I realise it is a different panel with possibly a different connector. Does your connector look like below?

[image]

doctordbx, 2020-05-31:
Well, my suspicions were correct. I connected everything up and dumped out the buttons and found they exist on a different BUS ID. I then hooked the panel up on the bench once I determined the voltage and can lines and found the panel behaves very differently indeed.

Going to need to spend a bit of time on this working this one out.

IDs
208 - Climate Control
2C8 - Stereo Controls
48F - Top button NAV switch and Phone button... but not Menu button.  Still looking for the menu button.

rhysmorgan134, 2020-06-05:
Hi, sorry I somehow missed this issue.

I believe they do look like the same IDs, except for the 48F one, I don't have the nav etc.

For mine, the power and home buttons go to the climate panel, the other lock, unlock, hazards go elsewhere.

As I replaced the panel, I hard wired these into the GPIOs (its nice that jag are switching 0v here as it means it can be tied straight to the input)

The best place for finding my IDs are in the following directory /electron/app/resources/canMap.json

`"520": {
        "2": [
            {
                "val": 1,
                "handle": "rearHeater"
            }
        ],
        "3": [
            {
                "val": 32,
                "handle": "auto"
            },
            {
                "val":64,
                "handle": "defrost"
            },
            {
                "val":128,
                "handle":"frontHeater"
            }
        ],
        "4": [
            {
                "val":32,
                "handle":"recirc"
            }
        ]
    },`

In the above example, the ID is 520 (this is the decimal value, hex value is 208), if byte 3, has the corresponding boolean for 32 active, then it says auto is turned on.

2C8 is also correct, I will check 48F, but if thats for nav and phone, I suspect that may not be on my vehicle

doctordbx, 2020-06-06:
Ugh of course, still... it didn't work so I went through and had another look at the codes.

So here's the codes I came up with:

    ID
    208 C0 [0A] 00 18 99 03 23 22 - Byte 1 - Sync Drv and Pax Temps (C0 sync 0A not sync) (state)
    208 C0 0A [02] 18 99 03 23 22 - Byte 2 - Rear Def 40 on 00 Off (pulse)
    208 C0 0A 02 18 [99] 03 23 22 - Byte 4 - Climate On / Off (state) 99 On / 19 Off
    208 C0 0A 00 18 99 [03] 23 22 - Byte 5 - Fan Speed. 00 Off 01-07 Speed 0F Auto (state)
    208 C0 0A 00 18 99 03 [23] 22 - Byte 6 - Driver Temp (20-39) (state)
    208 C0 0A 00 18 99 03 23 [22] - Byte 7 - Pax Temp (20-39) (state)
    2C8 00 [40] 00 00 00 00 00 00 - Byte 1 - Stereo On/Off (40 on)
    2C8 00 40 [10] 00 00 00 00 00 - Byte 2 - MENU button up top
    2C8 00 40 00 [40] 00 00 00 00 - Byte 3 - Mode Button Press (40 Pulse)
    2C8 00 40 00 00 [10] 00 00 00 - Byte 4 - Stereo Skip (10 Back 40 Fwd) (Pulse)
    2C8 40 40 00 00 00 00 00 7E - Byte 7 - Vol up / down (7E down 80 up)
    48F 00 3C 00 00 [40] 00 00 00 - Byte 4 - NAV button up top (need to wire this to GPIO)
    48F 00 3C 00 [40] 00 00 00 00 - Byte 3 -  PHONE button on top
    48F 00 3C 00 00 00 FA 00 [88] - Byte 7 MAX DEF  (on 88 off 00)
    48F 00 3C 00 00 00 FA [00] 00 - Byte 6 Recirc (on C0 off 00)
    48F 00 3C [00] 00 00 FA 00 00 - AV Button Press (01 On)
    48F 00 3C 00 00 00 [FA] 00 00 - This is a proximity sensor as far as I can tell (or some kind of light sensor) Value only changes if you put your hand near it, and it un dims the controls at night I think... it has been a while since I've actually driven the thing at night

doctordbx, 2020-06-06:
As you can see this is a bit different to yours and where yours send temp up and down pulses this one actually sends the temp. Likewise yours sends fan up or down this one sends the actual fan speed.

rhysmorgan134, 2020-06-06:
hmm, I think we need to look a bit deeper into it, IDs seems the same, I dont have a 48F, but the others are pretty spot on.

for 208, all of this data for me is indicators. So its not controlling anything, its feedback from what I assume is the climate control module.

So I do indeed have fan speed etc, but this ID only gives back the current set value, and it can't be used for controlling the speed, or defrosts etc.

Do you have the IDs for your fan up/down? from the speed knob?

doctordbx, 2020-06-06:
When I use cansend to send the 208 codes they definitely control the climate control. the only activity I saw on the bus when moving the fan dial was this:

208 C0 0A 00 18 99 [03] 23 22 - Byte 5 changing value corresponding with the fan

I couldn't see any activity on 2C8 when I adjusted the fan control.

It's 10:20 PM here now so can't go playing with the car unless I want to enter into a relationship status with all the time in the world to play with my car :-)

rhysmorgan134, 2020-06-06:
Sounds a no brainer!! ;)

If I get a chance I will look at making a seperate facelift branch, and try and implement some of those messages you sent over.

I need to double check my info, its been a while since messing about with the AC part so I could be telling you incorrect.

One thing if you do get a chance, would you mind comparing these settings I have in this video and seeing how your facelift compares?

https://youtu.be/o7TGF2G8eGw

doctordbx, 2020-06-06:
I saw that video pop up on my feed earier today and playing with the new software right now. Unfortunatley I have an Android Auto bug that is also present in Crankshaft that impacts my phone.

Your actual software on your Jaguar is a bit different to mine. A lot more colourful with different settings. Maybe I need a software update on the Jag.

rhysmorgan134, 2020-06-06:
I believe yours is the dark blue one? its the facelift, its much better than the version running in mine, mines an 08. Thats why it would be good to see if yours has the same settings or any extra.

doctordbx, 2020-06-06:
If you look at the top you can also see the plug for the back of my climate panel. I think this is different to yours?

doctordbx, 2020-06-06:
In terms of settings there's less there, although I think there are hidden menus that can be turned on. They like to hide things for the Australian market... not sure what that says about Australians.