rhysmorgan134/JaguarXf_CAN #12: refactor of code

rhysmorgan134, 2020-06-25:
Any issues from code refactoring to be captured here

rhysmorgan134, 2020-06-25:
@alierhan @doctordbx

Not sure if either of you are currently running the app, just a heads up, I've done a huge update, practically all of the code is rewritten, nothing should break, the only thing is that there needs to be 2 canbus channels now, can0 and can1. If you dont have a 2 channel hat, then you can create a virtual channel.

Also @alierhan I'm not sure if you ever resolved the brightness issue, but I will be putting a page in where you can see all of the live values, and scaled values, which will help.

alierhan, 2020-06-27:
Hi Rhys,

I am unable to run the app at the moment, dismantled my setup for this and using the Pi for another task on hand. I’ll be back at it later, in the meantime following the progress in great interest. I need to get the two channel can bus hat to really see the actual data, virtual channel is not going to provide any data.

The dimming issue still exists, scanned my car recently with iCarsoft apart from few errors resulting from replacing some of the T10 bulbs with LEDs there’s noting else. I now have a way of rebooting or shutting down the Pi without corrupting the SD card, I will get back to it when Pi is done it’s job.

Cheers,

Ali Erhan

Ali Erhan Photography
+61 414 760780
ali.erhan@me.com